Arcachofo/SimulIDE-dev #2: Issue with memory access

ThomasLeMezo, 2024-05-02:
https://github.com/Arcachofo/SimulIDE-dev/blob/304094d9865d0301397f159d5a913ba52c3c174a/src/microsim/mcudataspace.cpp#L103C1-L105C24

Bonjour,
Je crois qu'il y a une erreur sur cette fonction. Si elle est utilisée, cela bloque la mémoire puisque le `m_isCpuRead = true;` est après le return.

ThomasLeMezo, 2024-05-02:
Pour information, j'ai modifié le code de SimulIDE pour venir corriger automatiquement des épreuves d'électronique numérique avec une carte Arduino.
J'ai ajouté beaucoup de fonctions supplémentaires pour accéder aux différentes valeurs des composants, terminal série, vérification des connexions entre les composants, terminal série, etc.
Ce bug était assez compliqué à trouver ;)

Arcachofo, 2024-05-02:
Hi.
I don't see the problem, that value doesn't block anything.
The purpose of _m_isCpuRead_ is to detect when a RAM read is performed from "outside", for example from MCU Monitor.
In that case `m_isCpuRead = false;` indicating that it is not read from the CPU.
When RAM is read from the CPU then `m_isCpuRead = true;`

If your modifications are public, please let me know where I can have a look, maybe some of your changes could be useful.

ThomasLeMezo, 2024-05-02:
It seems that the line `m_isCpuRead = true` is never call as it is after the `return`.
There for, the function `isCpuRead()` always return false which makes the function `McuUsart::readByte( uint8_t )` not working in the case we perform previously a read on RAM.
I have to read the RAM to verify that the Arduino is configured by the program to the correct baudrate which requires to use the function `getRegAddress` with register `UBRR0L`.

I can send you a copy of my modified repo.

Arcachofo, 2024-05-02:
Thanks for the explanation, now I see it (maybe I need new glasses).
Hopefully solved at 824065c589cdce9207461576d5ee1d86340ffc3f

About your modifications:
I have in mind to implement some system to add some kind of test units to circuits or at least MCUs, so I'm curious about your use case and how you implemented it.
So if you want you can send to santigoro at mailfence dot com.

closed 2024-05-02 by ThomasLeMezo