Arcachofo/SimulIDE-dev #3: Numeric value written in the code with KY040

ThomasLeMezo, 2024-05-22:
https://github.com/Arcachofo/SimulIDE-dev/blob/f8a3f4ec82bacd54a4febdecfa7985d716f33028/src/components/micro/ky040.cpp#L153
In the case where the number of ticks is not 20, it might give wrong results?
we should have
```
m_dial->maximum()/m_steps - 1
```
instead of 19?

Arcachofo, 2024-05-22:
I don't think so.
20 dial steps is the size of one encoder cycle, when it is < 10 output is low otherwise output is high.
Note that m_dial->maximum() = steps*20

ThomasLeMezo, 2024-05-22:
If we change the number of steps per Rotation in the Properties, we don't have 400 (number of increments of the dial) / number of steps = 20. I might miss something . That was the case for 400/20 = 20.

By the way, it would be interesting to be able to connect the encoder to a motor.

ThomasLeMezo, 2024-05-22:
Ok, I see, you're right. You also set the maximum value of the dial inside the setSteps function.

By the way, it would be interesting to be able to connect the encoder to a motor.

closed 2024-05-22 by ThomasLeMezo

Arcachofo, 2024-05-23:
> By the way, it would be interesting to be able to connect the encoder to a motor.

Yes, that was planned, indeed I had the idea that it was already done.
Done at d69d2802c4dabea70ec83e968f7c633a0ea49e21
It works by linking a motor to an encoder:
Right-click in the motor and cloose "Link to Component", the click in the encoder.
To finish linking click in any empty place.