Arcachofo/SimulIDE-dev #4: Reload Firmware  and Reload Document

jmperricone, 2024-06-18:
First off all,   AMAZING !!!!!! Thank you Sr.  !!!
---

I like to use an external editor ( nvim ) and a makefile.

Checking "Reload hex at Simulation Start" or using "Reload firmware" dosen't work. 

I have to open the file in the editor, compile it and upload all from the app. Then i change the file in nvim, i have to close the file in the app editor, because Reload document dosen't work, open it again, compile and upload.

Thanks!!!

Arcachofo, 2024-06-18:
Hi.
It is working for me.

I think there is a problem with path.
If you compile and upload from simulide it will use a different path than compiling from nvim.
I think you should compile from nvim, then load that hex file from the MCU, then keep editing and compiling using the same path.

jmperricone, 2024-07-03:
It didn't happen again, and I couldn't make it fail again, so I closed this.

closed 2024-07-03 by jmperricone