Arcachofo/SimulIDE-dev #7: documentation: Implements traduction to "simulide_pt_BR.ts" file

Sergio-NA-prog, 2024-07-18:
# Contributing:

Any contribution is very welcome if it fits within the project plans.

To ensure that we are in synchrony and don't waste any time in vain, I propose this approach:

1. Propose.
- Traduce some words in English to Brazilian language. [file to traduce](https://github.com/Arcachofo/SimulIDE-dev/blob/master/resources/translations/simulide_pt_BR.ts)
3. Discuss.
4. Make a decision.
5. Work.
6. PR.

Arcachofo, 2024-07-20:
Hi.
Translations are usually done and the end of the cycle, when no changes in the texts are espected.
This is planned for around February-March 2025.

Sergio-NA-prog, 2024-07-20:
ok.. thanks

Arcachofo, 2024-11-15:
@Sergio-NA-prog Currently there is nobody in charge of Brazilian translation.
Do you want to take that role?

Sergio-NA-prog, 2024-11-16:
Hello.. I'm happy to be able to contribute

Sergio-NA-prog, 2024-11-19:
Hi. Almost terminated..

Arcachofo, 2024-11-21:
Thank you.
"Signal" on Scripted is not translatable and will probably be removed.

I created a discussion for translators to coordinate and discuss some points.
Have a look at: eeTools/simulide-translations -> Discussions

closed 2024-11-21 by Arcachofo

Sergio-NA-prog, 2025-02-05:
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE TS>
<TS version="2.1" language="pt_BR" sourcelanguage="en">
<context>
    <name>ADC</name>
    <message>
        <source>ADC</source>
        <translation>Conversor Analógico-Digital</translation>
    </message>
    <message>
        <source>Logic/Other Logic</source>
        <translation type="obsolete">Lógicos/Outros lógicos</translation>
    </message>
    <message>
        <source>Main</source>
        <translation>Principal</translation>
    </message>
    <message>
        <source>Size</source>
        <translation>Tamanho</translation>
    </message>
    <message>
        <source>Reference Voltage</source>
        <translation>Tensão de Referência</translation>
    </message>
    <message>
        <source>Electric</source>
        <translation>Elétrico</translation>
    </message>
    <message>
        <source>Other Logic</source>
        <translation type="obsolete">Outros lógicos</translation>
    </message>
    <message>
        <source>Timing</source>
        <translation>Tempo</translation>
    </message>
</context>
<context>
    <name>AboutDialog</name>
    <message>
        <source> Version:</source>
        <translation> Versão:</translation>
    </message>
    <message>
        <source>TextLabel</source>
        <translation>Título</translation>
    </message>
    <message>
        <source> Compiled:</source>
        <translation> Compilado:</translation>
    </message>
    <message>
        <source>About SimulIDE</source>
        <translation>Sobre o SimulIDE</translation>
    </message>
    <message>
        <source>&lt;a href=&quot;https://www.patreon.com/simulide/&quot;&gt;https://www.patreon.com/simulide&lt;/a&gt;</source>
        <translation></translation>
    </message>
    <message>
        <source>Website:</source>
        <translation>Página da Internet:</translation>
    </message>
    <message>
        <source>Patreon:</source>
        <translation>Modelo de Negócios:</translation>
    </message>
    <message>
        <source>Author:</source>
[... 5924 lines cut ...]
    <message>
        <source>GUI overload:</source>
        <translation>Sobrecarregado:</translation>
    </message>
    <message>
        <source>Real Speed:</source>
        <translation>Velocidade Real:</translation>
    </message>
</context>
<context>
    <name>manCompDialog</name>
    <message>
        <source>Name</source>
        <translation>Nome</translation>
    </message>
    <message>
        <source>ShortCut</source>
        <translation>Atalho</translation>
    </message>
    <message>
        <source>Manage Components</source>
        <translation>Gerenciar Componentes</translation>
    </message>
</context>
</TS>