Arcachofo/SimulIDE-dev #11: will multithreading be created?

Qikelela, 2024-08-04:
The program is not working fast enough. The program can use only 1 thread. You can try, for example, using OpenMP in for loops. I also recommend adding the -march=native flag when compiling.

Arcachofo, 2024-08-04:
SimulIDE uses 2 threads, one for the GUI and other for the simulation and I don't see a case for further multithreading.
If you know about a benefitial case for multithreading in Simulide please share your code.

Multithreading for loops can be faster in some cases, but it can also slow down the execution or not be possible at all.
-march=native can be used when you compile for your own machine, not to distribute sofware.

Qikelela, 2024-08-06:
The fact is that the program engine cannot reproduce complex schemes.

PaulaMaged, 2024-08-06:
How "complex" are we talking here?

- What are your computer/laptop specs?
- what circuits have you tried?
- what are the indicators that make you state that it is slow?
- Have you tried to see where the bottleneck might be? MCUs I believe are harder to simulate than circuit components

Your detailed insights will help me be aware of possible pitfalls as I begin to rely on this program and also help the developer to better pinpoint the root cause of the problem and discuss some possible alternatives/solutions for your case.

Qikelela, 2024-08-06:
> How "complex" are we talking here?
>
>     * What are your computer/laptop specs?
>
>     * what circuits have you tried?
>
>     * what are the indicators that make you state that it is slow?
>
>     * Have [...]

1.Computer specifications:
CPU: AMD Ryzen 7 5700x
RAM: 16 GB
GPU: nvidia GeForce GT 730 2gb
Xubuntu 22.04 64 bit

2. I built my processor from the old version 1.0.0.

3. Real Speed: 28%
    Engine load: 100%
    Update load: ~0%

Arcachofo, 2024-08-06:
> The fact is that the program engine cannot reproduce complex schemes.

Obviously there always will be circuits that will not run at real time, no matter how fast is the simulation engine.
But multithreading is not a solution in this case.

closed 2024-09-22 by Arcachofo