Arcachofo/SimulIDE-dev #17: segfault when opening compiler settings

blob42, 2024-10-12:
``Hi,

First thanks for this amazing project.  

I am getting a segfault when opening an .ino file and then opening the compiler settings window, it segfaults. 

I am running linux with Arduino IDE 2.3.3 and arduino-cli available.

I am not comfortable with C++ by a quick gdb stacktrace showd an issue with
[this line](https://github.com/Arcachofo/SimulIDE-dev/blob/master/src/gui/editorwidget/debuggers/inodebugger.cpp#L383).

I added a quick debug of that line and it shows:
```
------------------------------------------------- 
                   NEW CIRCUIT                    
 
[Thread 0x7fffea0006c0 (LWP 1813432) exited]
[Thread 0x7fffeaa006c0 (LWP 1813431) exited]
[Thread 0x7fffeb4006c0 (LWP 1813430) exited]
[Thread 0x7fffebe006c0 (LWP 1813429) exited]
[New Thread 0x7fffeb4006c0 (LWP 1814313)]
[New Thread 0x7fffea0006c0 (LWP 1814314)]
[New Thread 0x7fffebe006c0 (LWP 1814315)]
[New Thread 0x7fffeaa006c0 (LWP 1814316)]
enums:  "Uno,Mega,Nano,Duemilanove,Custom,;Uno,Mega,Nano,Duemilanove,Custom," 

Thread 1 "simulide" received signal SIGSEGV, Segmentation fault.
0x00000000004b8bd0 in QtPrivate::RefCount::ref() ()
```

- The command `aruindo-cli board listall` on my system shows
```
Board Name                       FQBN
ATmega1284                       MightyCore:avr:1284
ATmega16                         MightyCore:avr:16
ATmega164                        MightyCore:avr:164
ATmega32                         MightyCore:avr:32
ATmega324                        MightyCore:avr:324
ATmega644                        MightyCore:avr:644
ATmega8535                       MightyCore:avr:8535
Adafruit Circuit Playground      arduino:avr:circuitplay32u4cat
Arduino BT                       arduino:avr:bt
Arduino Duemilanove or Diecimila arduino:avr:diecimila
Arduino Esplora                  arduino:avr:esplora
Arduino Ethernet                 arduino:avr:ethernet
Arduino Fio                      arduino:avr:fio
Arduino Gemma                    arduino:avr:gemma
Arduino Industrial 101           arduino:avr:chiwawa
Arduino Leonardo                 arduino:avr:leonardo
Arduino Leonardo ETH             arduino:avr:leonardoeth
Arduino Mega ADK                 arduino:avr:megaADK
Arduino Mega or Mega 2560        arduino:avr:mega
Arduino Micro                    arduino:avr:micro
Arduino Mini                     arduino:avr:mini
Arduino NG or older              arduino:avr:atmegang
Arduino Nano                     arduino:avr:nano
Arduino Nano Every               arduino:megaavr:nona4809
Arduino Pro or Pro Mini          arduino:avr:pro
Arduino Robot Control            arduino:avr:robotControl
Arduino Robot Motor              arduino:avr:robotMotor
Arduino Uno                      arduino:avr:uno
Arduino Uno Mini                 arduino:avr:unomini
Arduino Uno WiFi                 arduino:avr:unowifi
Arduino Uno WiFi Rev2            arduino:megaavr:uno2018
Arduino Yún                      arduino:avr:yun
Arduino Yún Mini                 arduino:avr:yunmini
LilyPad Arduino                  arduino:avr:lilypad
LilyPad Arduino USB              arduino:avr:LilyPadUSB
Linino One                       arduino:avr:one
```

Arcachofo, 2024-10-12:
Thank you.
Solved at 4a8ddeb.

closed 2024-10-12 by Arcachofo

blob42, 2024-10-21:
Thanks @Arcachofo. It fixed the segfault but now the list of compilers on the "File Settings" dialog are shift down one position:

- I select Avrgcc:  log: Arduino Compiler successfully loaded.
- I select Arduino: log: AScript Compiler successfully loaded.

blob42, 2024-10-21:
I will add this comment here for future users to run SimulIDE with `arduino-cli` or `arduino-ide` on Arch Linux following these steps:

- Install `arduino-cli` (you can also install `arduino-ide`) package.
- create the folder `~/.arduino15/resources/app`
- create a symlink to the `arduino-cli` binary inside the created directory:
`ln -s $(which arduino-cli) ~/.arduino15/resources/app/`
- If the file `~/.arduino15/arduino-cli.yaml` does not exist run: `arduino-cli config init`
- Include the following yaml section if it is not already included:
```yaml
board_manager:
    additional_urls: []
directories:
    data: /home/spike/.arduino15
```

Now you should be able to load the Arduino compiler from "File Settings" and it will automatically detect all installed boards. Follow the `arduino-cli` official documentation for the rest.
        
You could add this to the FAQ or documentation page.