Arcachofo/SimulIDE-dev #53: Step over on Arduino doesn't work as expected due to commented line in source code

ancebfer, 2025-02-05:
Step over on Arduino doesn't work as expected.

This seems to be caused by this commented line in BaseDebugger::stepDebug method of basedebugger.cpp file:

            //if( m_functions.values().contains( PC ) )

Method source code fragment:
```
void BaseDebugger::stepDebug()
{
    if( !m_debugStep ) return;

    int lastPC = eMcu::self()->cpu()->getPC();
    eMcu::self()->stepCpu();
    int PC = eMcu::self()->cpu()->getPC();

    if( lastPC != PC )
    {
        if( m_over ){                                 // Step Over entry
            //if( m_functions.values().contains( PC ) )
            {
                m_exitPC = eMcu::self()->cpu()->RET_ADDR();
                m_over = false;
                if( PC == m_exitPC ) m_exitPC = 0;
                else return;
            }
        }
```
After removing the comment in this line, step over works right on Arduino.

What is the purpose of commenting this line?
As it affects BaseDebugger class, perhaps is it trying to solve some problem with other microcontroller architecture?

Arcachofo, 2025-02-06:
I need to have a look.
Probably it was just a test and I forgot to uncomment it again.

Arcachofo, 2025-03-12:
Line uncommented and fixed some other issues.

closed 2025-03-12 by Arcachofo