Arcachofo/SimulIDE-dev #54: Arduino avr-gcc tool path under MacOS

ancebfer, 2025-02-08:
Related to pull request #52, I have found that MacOS uses another path for avr-gcc tools:

- Windows: "C:\Users\<USER>\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\bin"
- MacOS: "~/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/"
- Linux: "~/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/"

This can be handled using QStandardPaths as:
```
// Windows: 
//  "C:\Users\<USER>\AppData\Local\Arduino15"
//   QStandardPaths::GenericDataLocation = "C:/Users/<USER>/AppData/Local"
m_toolPath = QStandardPaths::writableLocation(QStandardPaths::GenericDataLocation)+"/Arduino15";

// MacOS:
//   "~/Library/Arduino15"
//   QStandardPaths::GenericDataLocation = "~/Library/Application Support"
m_toolPath = QStandardPaths::writableLocation(QStandardPaths::GenericDataLocation)+"/../Arduino15";

// Linux:
//  "~/.arduino15"
//  QStandardPaths::HomeLocation = "~"
m_toolPath = QStandardPaths::writableLocation(QStandardPaths::HomeLocation)+"/.arduino15";
```
Source code of "InoDebugger::setToolPath" can be fixed as:
```
            if (m_toolPath.isEmpty()) // Fix empty config
            {
#ifndef Q_OS_UNIX
                // Windows
                m_toolPath = QStandardPaths::writableLocation(QStandardPaths::GenericDataLocation)+"/Arduino15";
#else
#ifdef Q_OS_MACOS
                // MacOS
                m_toolPath = QStandardPaths::writableLocation(QStandardPaths::GenericDataLocation)+"/../Arduino15";
#else
                // Linux
                m_toolPath = QStandardPaths::writableLocation(QStandardPaths::HomeLocation)+"/.arduino15";
#endif
#endif
            }
```
I don’t have a MacOS development environment, so THIS HAS NOT BEEN TESTED. This is why I opened an issue instead of submitting a pull request.

Arcachofo, 2025-02-08:
Thank you.
I will test and let you know.

ancebfer, 2025-02-09:
The reason of proposing Q_OS_MACOS instead of Q_OS_LINUX is to allow future support for other UNIX systems (like FreeBSD for example).

Arcachofo, 2025-02-09:
Ok, I see.
But Q_OS_MACOS is not working for me.
And I could not find one working, tried:
Q_OS_MACOS
Q_OS_MAC
Q_OS_OSX
Q_OS_DARWIN

ancebfer, 2025-02-09:
Q_OS_MACOS seems to be the recommended way in documentation:

https://doc.qt.io/qt-5/qtglobal.html#Q_OS_MACOS

But if it doesn't work, forget it (BSD support is not a priority at this moment).

Arcachofo, 2025-02-10:
Probably some problem in my configuration.
But yes, we can figure out later.

I commited the fix at: a752be7

closed 2025-02-10 by Arcachofo