Arcachofo/SimulIDE-dev #62: Compile error while with Arduino 1.8.1 which doesn't support -build-cache flag

hplc, 2025-03-31:
While compile my project, it will Executing:
`"C:/Program Files (x86)/arduino-1.8.1-windows/arduino-builder.exe" -compile -hardware "C:/Program Files (x86)/arduino-1.8.1-windows/hardware" -tools "C:/Program Files (x86)/arduino-1.8.1-windows/tools-builder" -tools "C:/Program Files (x86)/arduino-1.8.1-windows/hardware/tools/avr" -built-in-libraries "C:/Program Files (x86)/arduino-1.8.1-windows/libraries" -libraries "C:\Users\DELL\Documents\Arduino/libraries" -fqbn=arduino:avr:uno -build-path "C:/Users/DELL/AppData/Local/simulide/codeeditor/buildIno/build" -build-cache "C:/Users/DELL/AppData/Local/simulide/codeeditor/buildIno/cache" "C:/Users/DELL/Documents/Arduino/sketch_mar31b/sketch_mar31b.ino"`

Error report: flag provided but not defined: -build-cache

It seems arduino-builder.exe 1.8.1 doesn't support -build-cache!

I can manually run the compiler command  successed without -build-cache.

Checked source code, the flag hard coded in src/gui/editorwidget/debuggers/inodebugger.cpp, I think it should be configable through GUI or XML file.

Arcachofo, 2025-04-03:
Hi
Older Arduino versions are supported, but you can add another compiler if you want:
[https://simulide.com/p/compiler/#adding-compilers](https://simulide.com/p/compiler/#adding-compilers)

hplc, 2025-04-14:
Done with a custom compiler XML file, I especally need the hex file, so put then in D:/Arduino-build directory which need to be creaed firstly.

```<!DOCTYPE SimulIDE>

<compiler name="Arduino-1.8.1" type="arduino-1.8.1">
  <step 
    command="C:/Program Files (x86)/arduino-1.8.1-windows/arduino-builder.exe"
    arguments=' -compile -hardware "C:/Program Files (x86)/arduino-1.8.1-windows/hardware" -tools "C:/Program Files (x86)/arduino-1.8.1-windows/tools-builder" -tools "C:/Program Files (x86)/arduino-1.8.1-windows/hardware/tools/avr" -built-in-libraries "C:/Program Files (x86)/arduino-1.8.1-windows/libraries" -fqbn=arduino:avr:uno -build-path "D:/Arduino-build" $filePath'
  />
</compiler>```

![Image](https://github.com/user-attachments/assets/62d3c273-1bc9-4089-8764-712ef89c9f23)

closed 2025-04-14 by hplc