Arcachofo/SimulIDE-dev #63: [Bug] Wrong code path is passed to Arduino compiler

KelvinNovais, 2025-04-08:
## Information

OS: Fedora Linux 41
Package: Flatpak

## Bug Description

This problem is related to packaging the application in Flatpak format: since when SimulIDE opens files using the [FileChooser portal](https://docs.flatpak.org/en/latest/portal-api-reference.html#gdbus-org.freedesktop.portal.FileChooser), the app will receive an URI that points into the document portal fuse filesystem in `/run/user/$UID/doc/`. 

For example, when opening an Arduino code, its path will be represented by something like `/run/user/1000/doc/3a0497c5/code.ino`. I can edit the code without problems.

However, when trying to compile that code from the SimulIDE, the path passed to the compiler is not the real path outside the sandbox, which means the compiler can't find it, and returns the error:
```
Can't open sketch: main file missing from sketch: /run/flatpak/doc/3a0497c5/3a0497c5.ino
```

This problem may also happen with other compilers.

Arcachofo, 2025-04-09:
Hi.
I guess this is a problem with Flatpak, not Simulide.

KelvinNovais, 2025-04-10:
Yes, it's likely to be a problem with Flatpak. The file may be requested using portal, even if you don't do that explicitly.

By the way, is the Flatpak version officially supported?

lagodimos, 2025-05-16:
Hi @KelvinNovais! Could you check that the flatpak has the `filesystem=home` permission? (You can use Flatseal from flathub for that)

KelvinNovais, 2025-05-26:
The SimulIDE includes `filesystem=home` on its manifest, as you can see [here](https://github.com/flathub/com.simulide.simulide/blob/3980b36bb6b79929c141a4b81e40a1ef27f3c081/com.simulide.simulide.yml#L12)

The problem was that I was trying access files on an external device (my bad, sorry for the misunderstanding). Everything works fine on home directory.

closed 2025-05-26 by KelvinNovais