Arcachofo/SimulIDE-dev #67: STM32 Support?

lylythechosenone, 2025-09-10:
A quick search for "STM32" here and on the forum brings no results. Is STM32 support possible and/or being considered? It would be very cool, since most current STM32 simulators are proprietary.

Arcachofo, 2025-09-19:
I'm experimenting with Qemu to implement 32 bits MCUs that run at hundreds of MHz, by now only a very simple model for ESP32, but it's very experimental and there are a few challenges to solve.

It is possible to add support for STM32 (with Qemu) in Simulide but this stuff is still very experimental.
If someone is interested in doing it just let me know.

Arcachofo, 2025-09-20:
Looks like we are getting STM32 implemented:
[https://simulide.com/p/forum/topic/how-to-add-apm32-mcu-to-simuiide/](https://simulide.com/p/forum/topic/how-to-add-apm32-mcu-to-simuiide/)

We are already working in some basic implementation of STM32, from there APM32.